CIRDLES/Tripoli #178: Data visualization: Scale to fit block

noahmclean, 2023-10-26:
In "OnPeak Intensities by Time" tab for Preview/Review, the following behavior results in a plot that does not match menu choices:

With multiple species selected and plotted (e.g., all Pb isotopes for an NBS981 Pb analyses), right-clicking on a block and selecting "Sculpt this Block" results in correctly re-sizing and locking the x-axis limits. However, the data points plotted now only belong to the first species listed (e.g., 204Pb for that NBS981 analysis), even though all the other species' checkboxes remain selected.  

The correct behavior upon choosing "Sculpt this Block" should be that all the original menu settings (species, collectors, units, corrections, log-y-axis) carry over to the block view, with only the x-axis limits changing.  

[image]

bowring, 2023-10-26:
ALSO note that the intended behavior is that you can only sculpt one isotope at a time.

noahmclean, 2023-10-26:
Oh interesting... I would have guessed that you can sculpt any of the displayed isotopes, just so that you can visualize as much data at once instead of having to click through each isotope for each block to make any data rejection necessary.  

One use case for data rejection is that the ion source (filament or plasma/sample introduction system) takes an 'excursion' and produces a short time of bad/unusable data for all isotopes.  In this case, a user may want to reject the same one or ten or twenty seconds of data from all species.  

Drawing a box around that time period with all isotope intensities plotted would be a nice way to do this visually -- you could see when the excursion started and stopped.  The excursion may have a subtler influence on isotope ratios,  and therefore be harder to identify and reject from the "Ratios by Cycles and Blocks" tab.

bowring, 2023-10-26:
Two problems: 
1. sometimes the isotopes plot over each other such as 207 and 208 in "y-space"
2. more than one isotope has an intensity at the same time in "x-space", corresponding to the sequence table

One solution might be a button that synchs on time slots - any rejection rejects the others at the same time.

bowring, 2023-11-02:
see v0.3.6 soon!

closed 2023-11-02 by bowring