CIRDLES/Tripoli #180: Data visualization: rejection for whole analysis

noahmclean, 2023-10-30:
In "OnPeak Intensities by Time" tab for Preview/Review, make the data rejection/sculpting step faster by optimizing the mouse gesture actions.  

Anticipated most common actions from this tab, in order of frequency:
1. Reject a spurious outlier integration or small cluster of outlier integrations, identified by inspection.
1. Reject a bad block of data, identified by inspection.
1. Zoom in on a region of complicated behavior. The region could be ~1 cycle to ~2 blocks length, across one or more isotopes.
1. Zoom in for a block view to do more detailed data rejection.

Based on these common actions, I propose the following mouse gestures in the analysis view:
- Left-click drag to draw a red box.  When the left mouse button is released, all plotted data (checkbox is checked in the top menu) inside the box are either rejected or un-rejected.  The data inside the box are rejected if more than half of the data points inside the box are active (i.e. if the majority have not already been rejected)    The data inside the box are un-rejected if half or fewer of the data points inside the box are active (i.e. if half or more have already been rejected).  
- Right-click once anywhere inside a block to reject the entire block.
- Create a "zoom-in button" on the menu that changes the cursor from the default pointer to a magnifying glass with a little plus inside it.  Left-clicking with this new cursor performs a zoom-in action (both axes or just x-only or y-only if those menu checkboxes are checked).  Left-click-drag with this cursor draws a box and when the left mouse button is released, the boundaries of the box become the new [xmin, xmax], [ymin, ymax] of the plot.  Unpressing the "zoom-in button" will turn the cursor back to the default cursor and its behavior back to default behavior.
- Double-left-click on a block to zoom into the block view (currently "Sculpt this Block" view is available with a right-mouse click and selection from a pop-up menu).

This is a cool new new of mass spectrometer data, available nowhere else I know, so it is likely other users will have some input and opinions on how to navigate this window.

bowring, 2023-11-06:
With @noahmclean  inspiration, the Species Intensity Plot works like this:

There are two modes: **Selection** and **Sculpting**

Both allow any number of species to be shown
Both use scroll-zoom and RIGHT mouse pressed panning
Both allow block include/exclude with SHIFT + LEFT mouse click on block or block title

**Selection**: LEFT mouse draws a zoomBox that upon completion zooms in and can be repeated.
**Selection**: to ENTER **Sculpting** mode, LEFT mouse double-click
**Selection**: LEFT mouse double click on block name (at top) zooms into that block and enters **Sculpting** mode

**Sculpting**: LEFT mouse draws a selection box to toggle intensities as included / excluded (RED = excluded)
**Sculpting**: to EXIT **Sculpting** mode, LEFT mouse double-click

Excluded intensities are ignored in the log ratio calculations as shown in the Ensembles.

See upcoming version 0.3.7

bowring, 2023-11-06:
hope you like it

closed 2023-11-06 by bowring