CIRDLES/Tripoli #196: Export Measured and Modeled Data for Blocks

noahmclean, 2023-11-22:
Create a new exported CSV file to help test and diagnose issues with the data reduction and MCMC.  For each block, write a CSV file that contains the following eight columns (MATLAB variable names in parentheses):

1. time (d0.time)
2. measured intensities (d0.data)
3. measured intensity uncertainties (x0.Dsig)
4. isotope index (d0.iso_vec)
5. detector index (d0.det_vec)
6. baseline flag (d0.blflag)
7. cycle index (d0.cycle)
8. rejected true/false (d0.Include)
9. modeled intensities using the average of MCMC models (d = ModelMSData(x, d0))

All eight columns should be the same length.  If Tripoli handles the baseline and on-peak data separately, then they could be output in separate CSV files.

bowring, 2023-11-22:
do you really need baseline?

noahmclean, 2023-11-27:
Yes please!

bowring, 2023-12-02:
Provided in v0.4.2

closed 2023-12-02 by bowring