CIRDLES/Tripoli #214: Deal with Case1 data where some ratio data are <= 0

noahmclean, 2024-03-06:
Case1 data (OG Tripoli functionality) detects isotope ratio data.  For isotope ratio time series, Tripoli should take the logarithm of the ratios, evaluate statistics, then exponentiate.  This is the additive log-ratio transform of Aitchison's compositional data approach.  The problem is when an inaccurate baseline or deadtime measurement yields measured isotope ratios that are less than or equal to zero.   You can't take a logarithm of a negative number.

noahmclean, 2024-03-06:
Non-optimal solutions include: 

1. Taking the mean and standard error of all the isotope ratios as-is (no log transform, no rejection).  Downsides to this approach include (a) problem isotope ratios often include low-abundance isotopes with low count rates, and it's precisely these data that benefit most from the log-ratio transform, (b) relatedly, the mean will be biased towards a higher relative abundance of the low-abundance isotope, and (c) it's possible to end up with a negative mean result, even though you checked your mass spectrometer for antimatter leaks.

2. Reject the negative ratios, then evaluate the mean (or log-mean) of the ratios greater than zero.  Downsides to this approach include (a) and (b) above, but to a greater extent.  However, we avoid (c).

3. Add a small constant to each measured ratio so that the sum is $\geq 0$ for all ratios.  Downsides to this approach include (a) if the problem is a baseline or dead time subtraction, that subtraction will affect a collection of measured intensities equally but not necessarily a collection of measured ratios, each of which will include a different intensity, and (b) the potential to bias the ratio of interest by an amount up to that small constant (for arithmetic mean), or in an unpredictable way (geometric mean) depending in the latter case on the new minimum ratio value.

ryanickert, 2024-03-06:
ngl i've not checked for antimatter leaks lately.

ryanickert, 2024-03-06:
Q: If this is designed to accommodate inaccurate data, is it worth the effort?  Seems like suboptimal approaches might be fine if the data is not accurate in the first place.

Comment:  There may be cases where the data are accurate and still negative.  Very low precision (but accurate) analyses of baselines/backgrounds and very low precision (but accurate) analyes of ion beam intensites can yield negative values.

noahmclean, 2024-03-14:
A: You're right, we don't want to spend much time handling this specific scenario for legacy data.  Our new algorithms should improve the accuracy and precision of near-detection-limit measurements.  But, we still want to be able to handle older data files and formats, in part because we said we were going to.

So, non-optimal solution 1 above seems to be the thing to do.  This is (to my knowledge) the traditional way of handling this sort of data in the absence of direct intensity measurements.  The negative (but large uncertainty) ratios can then go off to data reduction software and be handled from there -- e.g., as an isobaric interference correction with a small magnitude but large uncertainty.

noahmclean, 2024-04-29:
Jim: The resolution to this question is the following:
Before calculating the statistics for an isotope ratio in Case1, check and see if any of its data points are $\leq 0$. If so, then treat the data like a 'User Function' -- take the mean, standard deviation, and standard error of the data without a log transform.  Report the statistics and shade the plot just like you a user function.  If the user rejects some data in 'sculpt mode', then check again to see if any data points are $\leq 0$ before re-calculating statistics.

bowring, 2024-05-13:
As of v0.5.2, the solution is to reject the negative ratios out of hand as per non-optimal item 2 above.  Will provide the "resolution" technique in a future release.

bowring, 2024-10-21:
@noahmclean's solution implemented - ratio will become user function in the presence of included negative or zero values. De-select them and user function returns to being a ratio.  v0.5.6

closed 2024-10-21 by bowring