CIRDLES/Tripoli #229: combining multiple data files

noahmclean, 2024-04-17:
Auto-detect and allow users to specify two or more files that should be concatenated.  Use time stamps (internal to the files rather than from the OS) to order the files in time.  Display the data together and calculate/export statistics across all concatenated data.  If one file contains a time series (e.g., an isotope ratio) that does not appear in another file, then that time series is imported where it exists and is blank/missing at times when the data is absent.  Blank/missing data do not affect statistics.

bowring, 2024-05-13:
example files please?

noahmclean, 2024-05-17:
Inside TripoliTestData/Workflows/ET_Redux U-Pb/GHR1-2020 find 
"GHR1 20200827-9 Pb" and "GHR1 20200827-9 Pb RUN2"
"GHR1 20200827-11 U" and "GHR1 20200827-11 U run2"

Inside TripoliTestData/IsotopxPhoenixTIMS/KU_IGL/IsolinxVersion2 find
"NBS981 230024a" and "NBS981 230024b" and "NBS981 230024d"

But under the rules above, you could combine any two data files.

noahmclean, 2024-05-29:
<do you mean for example combine "GHR1 20200827-9 Pb" and NBS981 230024b" ?>
Yes, this is a possibility, but would represent a pretty rare use case.  Seems like it should be possible, though, with the rules set out above.  

<If so, maybe we should call it combining analyses - load the analyses and then choose to combine the into a third "super" analysis?>
Combining analyses sounds great.  Spitballing here... 

For user input: Maybe there would be a way to drag/drop multiple analyses into the session editor, then multi-select the ones you want to concatenate?  Put a button below the "Session Analyses" box and above the "Notes" box that says <Concatenate>.  Pressing the button with multiple analyses selected draws a bracket around the analyses to concatenate them.  Whenever a user double-clicks on one of the analyses that are concatenated,, then all the concatenated analyses open together.

bowring, 2024-05-30:
question - check for same sample and fraction or user beware?

bowring, 2024-06-04:
v0.5.5 has an experimental concatenation feature for two sequential files with the same columns, walking before running.

Please provide input on naming, meta data, etc.  My first inclination is to remove the two (or more) target analyses and just leave the concatenated analysis.

Thoughts?

bowring, 2024-10-18:
I't been over 4 months and no-one has tried this ???

bowring, 2024-10-18:
@noahmclean - Since no one has used the feature since it was introduced for comment on June 5 almost 5 months), I am closing this issue.  If anyone wants to takeit further, please start a new discussion or series of issues.  Cheers

closed 2024-10-18 by bowring