CIRDLES/Tripoli #231: Copy summary statistics to clipboard

noahmclean, 2024-04-29:
A popular OGTripoli function is to copy/paste the 'Tripolized' means and percent standard errors of every plotted time series to the clipboard.  All numerical outputs should be the double precision number, output with of its decimal places (no rounding for output).  

For each time series, output:
<Name of First Time Series> (Mean, %StdErr)
<Mean>
<Percent Standard Error>
<Blank Space>
<Name of Second Time Series> (Mean, %Std Err)
<Mean>
<Percent Standard Error>
<Blank Space>
etc

For example,
i205 (Mean, %StdErr)
123456.22342394
2.3452323095789

204/205 (Mean, %StdErr)
0.0012323987235
0.5672350987234

206/205 (Mean, %StdErr)
1.2345678923497
0.0029232498723

bowring, 2024-05-10:
@noahmclean - we are not calculating or displaying %StdErr for non-ratio user functions - should that be true here?

noahmclean, 2024-05-10:
Let's keep them all consistent and provide %StdErr for all outputs. I think a mix would be confusing and easy to make mistakes with.  We will have more detailed options for tabular data outputs (e.g., csv files with more complete stats).

bowring, 2024-05-10:
Even though the plots don’t show them?

noahmclean, 2024-05-10:
Correct.

bowring, 2024-05-13:
done as of v0.5.2

closed 2024-05-13 by bowring