CIRDLES/Tripoli #235: Plot display on PC

Yashkincot, 2024-05-01:
Tripoli Version: 0.5.1
Java Version: 21.0.3
JavaFX Version: 21.0.3+1
OS: Windows 10 10.0

Issue details:
On a PC, the bars don't go over the xs in the reported statistics for the plots.  The red line for the mean extends beyond the left plot boundary. The block labels are not centered on the blocks on the x-axis.

[image]

bowring, 2024-05-02:
@Yashkincot - thanks for your thoughtful comments. The bar issue on windows is a font problem and the solution may be using a different font on windows.  The red line is by design to draw the user's attention - do you think it should protrude at all?  The block labels were designed to be left justified to denote the start of each block - do you think they need to be centered?
Cheers

joneelzinto, 2024-05-03:
Hey Jim, centering the block labels will be great, and truncating the red line at left plot boundary will also look better. 

Cheers.

commit ab6deb7a (2024-05-10): Merge pull request #237 from bowring/10may2024

Fixed #235 and related
closed 2024-05-10 by bowring