CIRDLES/Tripoli #240: New version comments and suggestions

YakovYermakov, 2024-05-14:
Tripoli Version: 0.5.2
Java Version: 21.0.3
JavaFX Version: 21.0.3+1
OS: Windows 11 10.0

Issue details:
1. When I scroll down, the graph zooms as expected, but when I scroll up, at some point the graph starts to shrink only on the Y axis (in zoom both mode), which can only be fixed by clicking "replot all" or by zooming only in Y axis mode.
2. It would be helpful to see a sample name at the top bar, as it was in old Tripoli.
3. X-axis caption "Blocks & cycles by time" looks redundant and eats up chart space.
4. I don't see options to close, hide or expand a specific graph inside "preview and sculpt data" window
5. See the image attached below:
[image]

bowring, 2024-05-14:
Thanks for your input.
1) requested by Noah - "x-axis should not show beyond x data"
2) good idea
3) good point
4) accomplished via the analysis manager "select columns to plot" tab
5) text spacing in blue is by design (alignment of decimals) - Noah's requirement, see https://docs.google.com/document/d/14PPEDEJPylNMavpJDpYSuemNb0gF5dz_To3Ek1Y_Agw/edit#bookmark=id.shth1rrfo2pa
expanding - need actual file example please
6) not sure what you mean , but Noah wanted legend to have large font

cheers. @bowring

YakovYermakov, 2024-05-14:
7. I can only move the chart window if I click and pull the left edge, there are only a few pixels to click on, and it works quite weird. See the video attached (sorry for the quality, GitHub only allows to upload 10mb)

https://github.com/CIRDLES/Tripoli/assets/130931450/7443d8fc-44d3-422e-8291-9a2025910882

YakovYermakov, 2024-05-14:
5. Here's my file for testing: 
[MD22-03B YYZ10B Pb.RAW.zip](https://github.com/CIRDLES/Tripoli/files/15311195/MD22-03B.YYZ10B.Pb.RAW.zip)
5.1 Because of the empty space on the left side, the "Legend" at the bottom is hidden, not sure if it is intended. Anyway, expanding the graph window doesn't really help, it overlaps by the window below. You can see it on the following screenshot: 
[image]
 
6. I meant error data on the left side, yes.  

Thank you for the quick response
Yakov

YakovYermakov, 2024-05-14:
8. Also, the number of digits after the decimal point differs for Y-axis values. Perhaps leaving zeros (e.g. 21.0 as in the 207/204 graph in the snapshot from 5.1) would be better for the view.

bowring, 2024-05-14:
7.  - FYI - The ability to move and resize plots was an experiment that never got finished.  It seemed Noah (only user before you) was maximizing Tripoli to fill the whole screen and show all the plots larger.  Also, deselecting certain plots frees up more real estate.  Then dbl click of plot area on right mouse maximizes the plot and another dbl click sends it back.  I think at this point, we need to develop a set of use cases for these plots to inform the next stage of development.

5.1 - maximizing the window (dbl right click on plot) provides enough room to show legend, etc.  Also maximizing Tripoli should make each plot big enough to show legend.  The info panel on the left does not rescale with the plot.

8. good point

@YakovYermakov - I am glad you are using Tripoli - this is the first detailed feedback I've had on the GUI.
I would also like feedback on the accuracy of the data reduction esp with regards to OGTripoli and outputs to Redux and clipboard.

Moving forward, I request that if possible, we use the [discussion forums](https://github.com/CIRDLES/Tripoli/discussions/239) for raising issues and then when ready, we make individual issues for each separate item.  (no need to rework this issue!)

YakovYermakov, 2024-05-14:
Thanks! 

For today, I tested Tripoli on a Windows laptop with only 13.3" screen. That's why I see these font size and spacing issues. On Mac with a big screen, it looks and works better.

Ok, I'm new in making reports on GitHub, will try to use discussion forums for suggestions and problems in the future.

Yakov

bowring, 2024-10-18:
@YakovYermakov - Let's close this issue and then, if any itemsare still outstanding, make onenew issue for each.  Thanks for your input!

closed 2024-10-18 by bowring