CIRDLES/Tripoli #247: Sculpt problems

ryanickert, 2024-05-20:
The attached file is a TIMSDP file but I renamed the file so it would upload.  When in tiled mode, the sculpt box will not follow the mouse and make it hard or impossible to sculpt Perc_BaPO4_on_202.

For Prec_Decrease I can only scuplt in one direction and only one cycle (of the low outliers) at one time.

Tripoli Version: 0.5.3
Java Version: 21.0.3
JavaFX Version: 21.0.3+1
[D218_Pb-6894.txt](https://github.com/CIRDLES/Tripoli/files/15378488/D218_Pb-6894.txt)

OS: Windows 10 10.0

Issue details:

ryanickert, 2024-05-20:
Looked at this in a bit more detail - it looks like the lack of ability to fully drag might be some sort of lag issue?  If I do it more slowly the box will expand.  I still can't sculpt the low early cycle from block 3 or the late cycle in block 6 (from Perc_BaPO4_on_202) when in tiled mode.  It works fine in full screen mode, so it might be related to the box boundaries not being equivalent to the block boundaries on the plot?  I also can't sculpt at all on Ratio 206/207 but I can on other ratios.  I can build the box but it won't reject any cycles.  Doesn't work in tiled or full screen mode.

noahmclean, 2024-05-21:
Thanks @ryanickert!  I've been able to reproduce your problem on my Mac running Tripoli 0.5.3.

bowring, 2024-05-24:
this will be fixed when we get selections across block boundaries like OGTripoli

PR #253 (2024-05-26): Various issues
Closes #228 
Closes #247 
Closes #248
Closes #249
Closes #250 
Closes #251
commit cc774f95 (2024-05-26): Fixed #251; fixed #228;  fixed #247
closed 2024-05-26 by bowring