CIRDLES/Tripoli #256: Tripoli Crashes on Triton .exp import

noahmclean, 2024-05-28:
Tripoli Version: 0.5.4
Java Version: 17.0.8.1
JavaFX Version: 17.0.8+1+b01
OS: Mac OS X 14.5

Issue details: Drag and drop opening a Triton TIMS data files (.exp files), e.g., from TripoliTestData, produces no response and the following error message in the terminal: 

Tripoli Resources loaded
Exception in thread "JavaFX Application Thread" java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
	at javafx.fxml/javafx.fxml.FXMLLoader$MethodHandler.invoke(Unknown Source)
	at javafx.fxml/javafx.fxml.FXMLLoader$ControllerMethodEventHandler.handle(Unknown Source)
	at javafx.base/com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventUtil.fireEventImpl(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventUtil.fireEvent(Unknown Source)
	at javafx.base/javafx.event.Event.fireEvent(Unknown Source)
	at javafx.controls/javafx.scene.control.MenuItem.fire(Unknown Source)
	at org.cirdles.tripoli.gui.TripoliGUIController.lambda$initialize$1(TripoliGUIController.java:196)
	at javafx.base/com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventUtil.fireEventImpl(Unknown Source)
	at javafx.base/com.sun.javafx.event.EventUtil.fireEvent(Unknown Source)
	at javafx.base/javafx.event.Event.fireEvent(Unknown Source)
	at javafx.graphics/javafx.scene.Scene$DnDGesture.fireEvent(Unknown Source)
	at javafx.graphics/javafx.scene.Scene$DnDGesture.processTargetDrop(Unknown Source)
	at javafx.graphics/javafx.scene.Scene$DropTargetListener.drop(Unknown Source)
	at javafx.graphics/com.sun.javafx.tk.quantum.GlassSceneDnDEventHandler.lambda$handleDragDrop$2(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at javafx.graphics/com.sun.javafx.tk.quantum.GlassSceneDnDEventHandler.handleDragDrop(Unknown Source)
	at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleDragDrop$11(Unknown Source)
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(Unknown Source)
	at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler.handleDragDrop(Unknown Source)
	at javafx.graphics/com.sun.glass.ui.View.handleDragDrop(Unknown Source)
	at javafx.graphics/com.sun.glass.ui.View.notifyDragDrop(Unknown Source)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
	at com.sun.javafx.reflect.Trampoline.invoke(Unknown Source)
	at jdk.internal.reflect.GeneratedMethodAccessor2.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
	at javafx.base/com.sun.javafx.reflect.MethodUtil.invoke(Unknown Source)
	at javafx.fxml/com.sun.javafx.fxml.MethodHelper.invoke(Unknown Source)
	... 36 more
Caused by: javafx.fxml.LoadException: 
file:/Users/n733m382/Documents/Tripoli/Versions/Tripoli-0.5.4.jar!/org/cirdles/tripoli/gui/AnalysesManager.fxml

	at javafx.fxml/javafx.fxml.FXMLLoader.constructLoadException(Unknown Source)
	at javafx.fxml/javafx.fxml.FXMLLoader.loadImpl(Unknown Source)
	at javafx.fxml/javafx.fxml.FXMLLoader.loadImpl(Unknown Source)
	at javafx.fxml/javafx.fxml.FXMLLoader.loadImpl(Unknown Source)
	at javafx.fxml/javafx.fxml.FXMLLoader.loadImpl(Unknown Source)
	at javafx.fxml/javafx.fxml.FXMLLoader.loadImpl(Unknown Source)
	at javafx.fxml/javafx.fxml.FXMLLoader.loadImpl(Unknown Source)
	at javafx.fxml/javafx.fxml.FXMLLoader.loadImpl(Unknown Source)
	at javafx.fxml/javafx.fxml.FXMLLoader.load(Unknown Source)
	at org.cirdles.tripoli.gui.TripoliGUIController.launchAnalysesManager(TripoliGUIController.java:449)
	at org.cirdles.tripoli.gui.TripoliGUIController.manageAnalysisMenuItemAction(TripoliGUIController.java:462)
	... 46 more
Caused by: java.lang.NullPointerException: Cannot invoke "org.cirdles.tripoli.sessions.analysis.massSpectrometerModels.dataSourceProcessors.MassSpecOutputBlockRecordFull.baselineIDs()" because the return value of "java.util.Map.get(Object)" is null
	at org.cirdles.tripoli.sessions.analysis.Analysis.prettyPrintAnalysisDataSummary(Analysis.java:484)
	at org.cirdles.tripoli.gui.AnalysisManagerController.populateAnalysisDataFields(AnalysisManagerController.java:358)
	at org.cirdles.tripoli.gui.AnalysisManagerController.populateAnalysisManagerGridPane(AnalysisManagerController.java:319)
	at org.cirdles.tripoli.gui.AnalysisManagerController.initialize(AnalysisManagerController.java:251)
	... 56 more

bowring, 2024-05-28:
which file?  .exp work for me ( i didn't try all of them)

bowring, 2024-05-28:
when it crashes like that, restart it

noahmclean, 2024-05-29:
Sent you a DM

bowring, 2024-05-29:
Thanks.  It turns out  a few of the Triton files have empty lines at the end labeled with a cycle number and that was not considered in the parser.

Fixed in v0.5.5

commit 620173de (2024-06-04): Fixed #256