CIRDLES/Tripoli #262: Blue box glitch

YakovYermakov, 2024-06-26:
Tripoli Version: 0.5.5
Java Version: 21.0.3
JavaFX Version: 21.0.3+1
OS: Windows 11 10.0

Issue details:

Hi Jim! 

Found a weird glitch when zooming all plots (see screenshot). It happens after zooming in (by mouse left and drag) and then trying to zoom all plots (ctrl+scroll). Even if you zoom back it and then zooming all the plots this glitch appears.  It disappears if you click "replot all"
[image]

Seems like only a Windows problem. I tried to do the same on the Noah's Mac, and it didn't happen.

bowring, 2024-07-29:
@YakovYermakov - Please run from terminal in verbose mode and send me everything when this happens.

YakovYermakov, 2024-08-16:
@bowring Hi Jim, we checked with Noah, it seems like verbose mode doesn't work at all, neither on Mac nor on Windows. The command "java -jar Tripoli-0.5.5.jar -verbose" launch Tripoli, but it makes no recordings in terminal.

bowring, 2024-10-18:
@YakovYermakov - I can't duplicate it on windows.  Please supply the exact file in case this is related to a data anomaly unique to it.  Thanks!

bowring, 2024-11-10:
@YakovYermakov - is this fixed?