CIRDLES/Tripoli #263: Uncertainty bars

YakovYermakov, 2024-06-26:
Hi Jim!
I think that these error bars aren't working. I mean, the height of the bar should have any sense, right? Or they represent only the color legend? 

Also, please check bar labels. 2sigma x bar should be the green one.

[image]

noahmclean, 2024-06-27:
To clarify, @YakovYermakov means the heights of the red, yellow, and green bars above the red line in the legend in the lower left of the screenshot.  The green bar is the same height as the yellow bar in the legend but is generally narrower than the yellow shaded area in the plots.

bowring, 2024-06-27:
this is OGTripoli:
[image]

bowring, 2024-10-18:
fixed V0.5.6