CIRDLES/Tripoli #276: Concatenation

riouxm, 2024-10-28:
Hi Jim,

I just tried out the concatenation and it is not currently working. It does produce a concatenated file, but I am not able to plot the data from that file. I have attached the two data files that I was trying to concatenate.  

I'll generate a second post with some ideas for concatenation functionality.

Matt

[B1551_F09_230913M01 z18-1 Pb-3-51702.xls](https://github.com/user-attachments/files/17549229/B1551_F09_230913M01.z18-1.Pb-3-51702.xls)
[B1551_F09_230913M01 z18-1 Pb-51700.xls](https://github.com/user-attachments/files/17549231/B1551_F09_230913M01.z18-1.Pb-51700.xls)

bowring, 2024-11-02:
Fixed in v0.5.8

bowring, 2026-02-22:
see v2.2.1 for first two use cases