CIRDLES/Tripoli #279: .trip files

riouxm, 2024-11-04:
Hi Jim,

I was wondering if you are planning to make it possible to open .trip files in the new Tripoli? This would be very useful when working on datasets that were originally reduced using the older Tripoli and saved as .trip.

Thanks.

Matt

bowring, 2024-11-05:
Great idea, but not practical.  However, if you designed a schema of all the data you would like to in the new Tripoli, we could export it from the old Tripoli and read it with the new Tripoli.

riouxm, 2024-11-06:
I am not sure that is a practical workaround, as many labs have thousands of old .trip files. Would it be possible for new Tripoli to extract the data from the .trip file, then import the data into new Tripoli?

For my purposes, I can just import the raw data files and generate new Tripoli files, but if the goal is to have new Tripoli replace old Tripoli, it would be nice to have a way to view old .trip files.

ryanickert, 2024-11-06:
I agree with Matt that this is a problem for the (evenutal) uptake of NuTripoli.  For workaday U-Pb ID-TIMS data, OG Tripoli works so well that most labs will be reluctant to move away from it if there is no backwards compatability.  If that's the case, our lab would most likely continue to use OG Tripoli until it catastrophically breaks and only use NuTripoli for peculiar, bespoke analyses.

bowring, 2024-11-06:
Unfortunately, it is not a practical possibility to make it backwards compatible.  However, my proposal is only 2 clicks longer:
1) drag .trip onto OGTripoli
2) click the new "Export button" to create a .tripx file
3) drop .tripx file onto Tripoli and see your results

The above could be automated as a batch file.

If this is to work, a schema needs to be developed as per above that details exactly what info should be exported.

Cheers