CIRDLES/Tripoli #280: Number of ratios

riouxm, 2024-11-06:
Hi Jim,

Tripoli is incorrectly interpreting the number of ratios per block for BSU data. With their run parameters, the ratios per block is set to 15, but you end up with 16 datapoints per block, due to beam interpolation. Currently, Tripoli brings everything in as 15 ratios per block. I thought I could correct that using the "Set default cycles per block" in Tripoli, but that does not seem to be working.

I have attached an example data file.

Thanks.

Matt

[891_F19_735B_1271m iz18 Pb-8932.xls](https://github.com/user-attachments/files/17652245/891_F19_735B_1271m.iz18.Pb-8932.xls)

ryanickert, 2024-11-06:
Is this a triple vs. double interpolation issue?

noahmclean, 2024-11-06:
I think it's just a matter of what IonVantage says (15 ratios per block in CTRL sheet, cell ILV_CyclesPerBlk) vs. what it does (the value of that cell plus one if CRTL sheet cell ILV_ApplyBeamInterp = TRUE).

riouxm, 2024-11-06:
My sheet does show 15 cycles per block with Beam Interpolation = True, so maybe this is a bug? Old Tripoli does get it right.

noahmclean, 2024-11-06:
Yeah, I vaguely remember this.  Your blocks all show as having 16 ratios in this file, which is what we need to read in from the cycle tab.  I think for OGTripoli, we just checked the `ILV_ApplyBeamInterp` cell on the *CTRL* sheet and, if TRUE, added one to the value in the `ILV_CyclesPerBlk` cell.

bowring, 2024-11-10:
Currently, the feature allowing for change of default cycle count only works if no count is found, which explains why switching to 16 did not work.  Let's keep that functionality if all agree.  The beam interpolation miscount is fixed separately.
Fixed in v0.5.8.